inventables/x-carve-instructions #21: A Schematic of the wiring to the powersupply and G-Shield would help those unfamiliar with wiring

Casadelwhacos, 2015-06-11:
I found this while searching the forums and thought it might help those unfamiliar with wiring

[image]

paulkaplan, 2015-06-23:
@chrisbalin what is the status of this?

vaporlock, 2015-08-04:
+1 for this. This one image replaces the whole page. The instructions don't even list the pin numbers. We are forced to count black holes in a black connector. The orientation of the connector is also not specifically called out as a caution. Anytime something can be hooked up backwards it will be. The limit switch ground wires don't fit well in the board ground connector and should either be spliced in before the screw down or connected to the PS ground, which is on the same wire. Is noise a potential reason not to do this? If so, better to tie them to the actual ground pin on the board rather than power ground.

chrisbalin, 2015-08-06:
@johndudleyhayes Were you going to make your own version of this diagram?

closed 2017-11-28 by griffindennis