inventables/x-carve-instructions #22: Translating the instructions to different languages

ZachKaplan, 2015-06-14:
We had a customer ask if we could translate the instructions to Spanish.  I found that Google Translate offers this service but you need to add this code snipped to each page you want translated.

<div id="google_translate_element"></div><script type="text/javascript">
function googleTranslateElementInit() {
  new google.translate.TranslateElement({pageLanguage: 'en', layout: google.translate.TranslateElement.InlineLayout.SIMPLE}, 'google_translate_element');
}
</script><script type="text/javascript" src="//translate.google.com/translate_a/element.js?cb=googleTranslateElementInit"></script>

paulkaplan, 2015-06-23:
I thought google translate automatically offered to do that when using chrome?

ZachKaplan, 2015-06-23:
According to their instructions only if you add the code snippet.

-- 
Zach Kaplan
CEO

Inventables
600 W. Van Buren #602
Chicago, IL 60607

o 312.775.7009
m 773.936.4975
f 413.332.0054

w inventables.com
t http://twitter.com/inventables

The Hardware Store for Designers - Igniting the Digital Manufacturing
Revolution

closed 2017-11-28 by griffindennis