inventables/x-carve-instructions #25: Spindle mount step doesn't make alternatives clear

paulkaplan, 2015-06-23:
The spindle mount step goes directly into the 24v spindle option without mentioning that there are 2 other possible spindle mounts. Can we use the collapsable accordion to make this difference more clear? Or at least a table of contents?

This is the step I'm referring to: http://x-carve-instructions.inventables.com/step12/

chrisbalin, 2015-06-23:
The subnav on the right serves as a table of contents. I added a note to make it more clear.
[image]

Do you think this resolves this issue?

tleswing, 2015-06-23:
I don't think there shouldn't be a default. I think the whole thing should
accordion.

-- 

Tait Leswing

Project Experience Manager

Inventables

600 W. Van Buren #602

Chicago, IL 60607

o 312.775.7009

m  414.943.3343

f   413.332.0054

w  inventables.com   <http://inventables.com/>

t    http://twitter.com/inventables

The Hardware Store for Designers - Igniting the Digital Manufacturing
Revolution <http://www.youtube.com/watch?v=fiUEC_TQJ-Q>
YouTube - Videos from this email

paulkaplan, 2015-06-23:
I think if you have to "point" to the table of contents, it is in the wrong place.

chrisbalin, 2015-06-23:
Yeah I will make them into accordions like @tleswing suggested.

chrisbalin, 2015-06-24:
Can I mark this resolved now? This is what the page looks like now:
[image]

closed 2015-06-24 by paulkaplan

tleswing, 2015-06-24:
Looks great

-- 

Tait Leswing

Project Experience Manager

Inventables

600 W. Van Buren #602

Chicago, IL 60607

o 312.775.7009

m  414.943.3343

f   413.332.0054

w  inventables.com   <http://inventables.com/>

t    http://twitter.com/inventables

The Hardware Store for Designers - Igniting the Digital Manufacturing
Revolution <http://www.youtube.com/watch?v=fiUEC_TQJ-Q>
YouTube - Videos from this email