inventables/x-carve-instructions #57: Flanged Bearing Protrudes Past Z-Axis Plate.

LeviMThomas, 2016-12-12:
The flanged bearing protrudes past the back of the Z-Axis plate causing the Z-Axis rail not to seat flush or square with the Z-Axis plate. Also on mine one of the holes in the Z-Axis rail was crooked and unfixable. 
[image]
[image]

LeviMThomas, 2016-12-12:
Im an idiot. lol. I had the Z-Axis rail on the wrong side of the Z-Axis plate.

closed 2016-12-12 by LeviMThomas

griffindennis, 2016-12-12:
No worries- I was scratching my head there for a second myself. If you have any more issues you can always send in a case through our support center or give us a call at 312-775-7009. Thanks, -Griffin.

LeviMThomas, 2016-12-12:
I have another issue I can’t seem to find a solution to. The holes for the eccentric spacers on the X-Carve DeWalt 611 Spindle Mount seem clean on the instruction photos, allowing them to rotate to make adjustments where as mine has tapered holes. So the hexagon part of the spacer goes down into the tapered hole locking it in place and causing one of the V wheels to be out of square no matter how it put it. I have tried swapping the two wheel assemblies(bolts and all) and either way the bottom one is crooked like that. Is the hole supposed to be tapered like this?

LeviMThomas, 2016-12-13:
Hello again. 
I just unboxed everything today in a completely empty room in my house were I am assembling this xcarve and have been diligent about keeping all of these parts organized. I got to the Belting section where it calls for 16 of the 25286-33 - Button Head Cap Screw M5 x 8  and there are only 6 left in my bag that is labelled Qty 16. Thinking I had been using the wrong screws I went back to the instructions and previous steps had in fact called for the 25286-33. So thinking there may have been another bag I went through all of my empty bags and none of them were labeled 25286-33 nor do I have 10 of any other screw that I could have mistakenly used. Was there supposed to be another bag of those screws? I am becoming less than impressed as this is the third thing so far that has been defective or missing and Im only half way through the build. 

Levi Thomas
4112 Woodland Ave
Kansas City MO, 64110

Thanks,  

Levi Thomas 
913-940-6116
Owner - RevampKC.com <http://revampkc.com/>