inventables/x-carve-instructions #91: Dust Control Instructions and Homing Switches

bible9230, 2017-10-13:
The dust control instructions need to mention a homing switch relocation.  There is a good chance that their existing homing switch location will allow the dust control shoe and brackets to collide with the y axis end plate.  

The location should be about 1" from the Y plate and 2" from the X plate.
[image]

[image]

griffindennis, 2017-10-13:
Good call, This is definitely something that needs to be changed. This will be taken care of when integrating the dust shoe instructions themselves.

griffindennis, 2018-01-31:
Added note on this in d4aab767260e1662aa4888a3546c2118b8e35bb6

closed 2018-01-31 by griffindennis