inventables/x-carve-instructions #139: Acme Lead Screw test/wobble

pcstolz, 2019-07-11:
We often get reports of a bent lead screw at these steps of the instructions:

http://x-carve-instructions.inventables.com/1000mm/step2c/4leadscrew/
http://x-carve-instructions.inventables.com/1000mm/step2c/5attachtocarriage/

This is before the mount is attached to the carriage, so the lead screw is spinning freely at the bottom giving a 'wobbling' appearance that leads many people to think there is an issue with the screw or the bearing.

I would suggest adding a step to test the lead screw on a flat surface before attaching the lead screw, as well as a line stating that the lead screw may appear to wobble until it is anchored to the spindle mount.

johndudleyhayes, 2019-07-16:
@pcstolz I'm sort of afraid of suggesting to test on a flat surface just because we can't verify the flatness of that surface.  do you think simply stating that a wobble might appear (because it's only secured at the top of the spindle mount) would be enough to mitigate these types of cases?

pcstolz, 2019-07-16:
Yeah that's fine - Julie had suggested the flat surface thing

-- 

Paul Stolz

*Sales Support Manager*

Inventables

328 S. Jefferson St, Suite 260
<https://maps.google.com/?q=328+S.+Jefferson+St,+Suite+260+Chicago,+IL+60661&entry=gmail&source=g>

Chicago, IL 60661
<https://maps.google.com/?q=328+S+Jefferson+Street,%C2%A0+Suite+260+Chicago,+IL+60661&entry=gmail&source=g>

Make Your Mark

Launch your business with X-Carve <https://youtu.be/CyMNIBAMbRY>