inventables/x-carve-instructions #155: Customer Feedback - Sideboard and Drag Chain Sections

johndudleyhayes, 2019-09-23:
> You build the frame, attach the board and then attach it to the x-carve. What I didnt see was anything mentioning the offset for the base between the side board and x-carve. In the beginning for the x-carve it was explained that the 30mm was important for later but never told why. Now I know but it cost time.  

> Another example was trying to figure out where to attach the drag chain bracket on the y-axis which wasn't shown or explained where it goes but having softer through some later pictures it showed the bracket. But the picture almost made it look like it went under the x-carve and screwed in from the inside of the frame not the outside.

giovannadf, 2019-10-16:
Picture after "Wire up z-axis and x-axis stepper motors and limit switches..." shows screwing the drag chain but does not call out which screws. 

[image]
Call out what screws are being used on the picture above. Line can read: "Attach the drag chain end to the drag chain bracket on the x-carriage using two M5 x 12mm Flat Head Cap Screws and two M5 Nylon Insert Lock Nuts. "

giovannadf, 2020-02-21:
Customer feedback: 

> The picture below: “Assemble frame using the 20mm x 20mm aluminum
> extrusion, gusset brackets, pre-assembly insertion nuts and M5x10mm
> button head cap screws."

Shows a rectangular frame. The next picture shows a rectangular frame with the
board sitting in between the long extrusions. It seems to use four M5x12mm button
head cap screws. BUT the next picture shows the frame build differently. It shows
an extension extending past the end of the rectangle. The next picture actually
shows that there is another M5x10mm screw, and the right, long extrusion should
actually be offset left from the edge. No measurements given.

>Insert the two post-assembly t-slot nuts and slide into place, lining them
>up with the holes in the bracket, and secure the drag chain end in place.

This is the first picture (that I could find) showing the Drag Chain Bracket. I think it shows the bracket pointing down. I still can not tell where to locate the bracket. The important part of the next picture is too dark.

Resolve by calling out frame does not need to be tightened as it will be adjusted to fit under sideboard. Also, call out which screws holes should line up with aluminum extrusions/t-slot nuts.

giovannadf, 2020-02-21:
Customer Feedback: 

> Attach the drag chain bracket to the aluminum extrusion frame using
> the M5x6mm button head cap screw and post-assembly insertion nut.

I can’t tell where to attach this bracket. After searching all the pictures, I was able to
find an assembled picture that hinted at where this bracket goes. I also can’t tell if it
is right-side-up, or down.

Resolve the issue by taking a new image of positioning or changing step to optional if you do not plan to add sideboard.